freetserv/freetserv #26: enhancement: create a header for the ethernet port

stapelberg, 2015-12-06:
We can just not populate it on 3 of the 4 boards.

jaredmauch, 2016-01-03:
I can tackle this one.  It's super simple.

stapelberg, 2016-01-04:
Great, looking forward to your pull request (please keep the changes as small as possible). Let me know if you need anything.

jaredmauch, 2016-01-04:
Well, I haven't completed it, but thought I'd ask your feelings on this placement first before I fix all the impacted traces, length matching, etc.

[image]

stapelberg, 2016-01-04:
Thanks for the preview. I think the general position is fine, especially since the daisy-chain connector on the right hand side does not need to be populated for the right-most board.

Here are a couple of thoughts:

* Is the part that you’re adding available on Digi-Key?
* Can you come up with a way to route it without touching any other traces to make this change as non-invasive as possible? It looks to me like it might be possible to use traces on the top layer that go from top to bottom on the right hand side of the PCB, all the way to P12.

jaredmauch, 2016-01-04:
Yes, it's available on Digikey (WM5467-ND) and there are other variants which are more expensive or cheaper, eg: WM5468-ND which are pin compatible.  It's not really possible without moving part D2, which I assume is ok.  The traces related to D2 have to be re-routed.  This would obviously impact anyone using a PnP machine to place parts.

Is there a common .lbr file for freetserv I should put these parts in, or should that go in the pull?  I could also just email you the files offline.

stapelberg, 2016-01-04:
Moving D2 is okay. We don’t have a common .lbr file yet, but you’re welcome to start one, or just let Eagle embed the lbr file into the brd/sch files as appropriate (which is what we currently do).

jaredmauch, 2016-03-03:
taking a fresh look at this, i'm concerned that having the ethernet pin 5 shared with the RS232 ground may cause signal issues, have you had issues with the soldered cable connection being shared with the uart12 level shifter?

jaredmauch, 2019-12-18:
since this was merged, should this be closed?

stapelberg, 2019-12-19:
Yep, thanks

closed 2019-12-19 by stapelberg

paravoid, 2020-12-20:
I've noticed that the documentation (assembly.html) still documents a manual way, and references this issue as a work-in-progress. Not sure how this should be updated though.

stapelberg, 2020-12-20:
@jaredmauch Any thoughts on this? Personally, I don’t have a build with the new ethernet port solution, and don’t plan on building one right now.