freetserv/freetserv #35: nit: find correctly colored cable (green/yellow) for grounding

stapelberg, 2015-12-24:
Currently, we use Digi-Key A119543-ND, but that doesn’t have the right color. Find a cable with the correct size and color.

athompso, 2019-07-15:
As of 2019jul15, DigiKey has 38 options to choose from that are "Normally Stocking":  https://www.digikey.com/products/en/cables-wires/single-conductor-cables-hook-up-wire/474?k=&pkeyword=&sv=0&pv1101=3&pv1101=5&sf=1&FV=1380005%2C1480037%2C1480227%2C1480039%2C504001a%2Cffe001da%2C11340003%2C11340005%2C11340006&quantity=&ColumnSort=0&page=1&nstock=1&pageSize=25

The problem is that they all start at 100' (30m).

I checked Mouser, and they have the same problem - lots of options https://www.mouser.ca/Wire-Cable/Hook-up-Wire/_/N-5ggsZscv7?P=1y8fm3rZ1z0xf08Z1y8twnyZ1yzoqgaZ1z0xf5nZ1y8pui4Z1z0vstfZ1y8trjmZ1yx79kqZ1z0x8gx&Ns=Pricing|0 assuming you want 100' or more.

Any local electronic or auto-parts supply store will have similar options; in the auto industry this is "primary wire", and often comes in rolls as small as 25'.  Or, just ask your local mechanic for a length of 18ga, green primary wire and it'll probably be free.  (Note that primary wire is often solid copper, though, not stranded.)
Higher gauges are usually OK for a grounding wire, and that increases your options, for example anywhere in Canada you can order this ridiculously cheaply: https://www.princessauto.com/en/detail/16-gauge-25-ft-green-primary-wire/A-p8216475e

If in the USA, you can find this nearby: https://www.homedepot.com/p/Southwire-25-ft-16-Green-Stranded-CU-GPT-Primary-Auto-Wire-56422021/202316507

I've no idea what to get in Europe, but I'm sure something similar exists - certainly mechanics and electricians are not noticeably more rare there than here in Canada, so shops must exist to service them.

I hate to encourage bad practice, but it would probably be cheaper and easier to destroy a standard PC power lead and scavenge 3ft of green-jacketed, stranded-copper, ground wire from that :-(.

athompso, 2019-07-15:
Also viable, but PITA: wrap colored electrical tape around the wire and/or use green heat-shrink tubing.

stapelberg, 2019-07-16:
Thanks for the details, that’s very helpful!