freetserv/freetserv #37: Case redesign suggestions

alexandrujuncu, 2015-12-29:
Hi!

Just throwing in a couple of cents on the hardware design, if you find it helpful:
1) From what I understand from the pictures, it looks like the first RJ45 port is actually an Ethernet port while all the others are serial ones. This is both confusing and dangerous if somebody confuses the ports might end up burning the port (serial and Ethernet are not electrically friendly and I've seen a lot of people destroying ports because they plugged in the wrong one).
I think the Ethernet port should be clearly separated from the others. Also it might be better to NOT use an extender and just have the RJ45 port from the RPI directly accessible. Which brings me to the next point...
2) The Raspberry Pi board should not be the middle of the box, but at the edge accessible to the user. This way the side with the RJ port and the 2/4 USB ports would be accessible to the use. The USB ports could be used for miscellaneous things for the user so why not expose them?
But where to pace it? It will take up some room in the front. So the logical solution is instead of the power plug. And, again, this brings me to another point:
3) The power plug should not be in the front. I have seen many racks with many routers, switches and servers and with very few exceptions (I remember some annoying Cisco ASA/PIX-s) equipment has the power plug in the back. As it should. Power is provided through the back to be consistent with the rest of the equipment and also not block rack door from closing (power plugs tend to take up more room).

So, in conclusion, I suggest you put the Raspberry Pi on the left of the console ports, in the front. This way you would separate the Ethernet port from the serial ports and have direct access to the Ethernet and USB ports on the board itself. The big power plug could and _should_ be put in the back where it belongs and not get in the way of anything else.

Hope this helps. Thanks!

stapelberg, 2015-12-29:
The front side of freetserv has the serial ports and the power plug on the same side because servers do the same thing (see e.g. https://images.stockinthechannel.com/pic/-pie3aUTqUmBmdYcnONR9A.c-o.jpg). This leads to all the cables being on the same side of the rack, i.e. less clutter and less long cables.

If one wants to go for a case where some parts are on the front side and some are on the back side, I think it’d be reasonable to place the Raspberry Pi’s ports facing the back side, so that one gets the benefits of having the Ethernet port directly accessible that you mentioned. One tiny side node on the USB ports, though: freetserv is intended to be a single-purpose device, so there shouldn’t be a need to use the USB ports, or at least not in any user-facing fashion.

In general, I don’t have strong feelings on the way the case is designed. The current design works very well for the use-case that I’m solving for myself, and I’m publishing the files because they’re tested and found working. If other people have other use-cases/preferences, they’re welcome to modify the fpd files before ordering the case (or build an entirely new case altogether), and I’m happy to accept these modifications to live in the same repository. The only thing I ask is that such modifications are actually tested, and not just theoretical :).

I’ll close this issue as nothing remains to be done, at least not from my side. Thanks for your input!

closed 2015-12-29 by stapelberg

alexandrujuncu, 2015-12-30:
Hi!

That image is actually actually backs the idea that power plugs should be at the back :-?

I've never worked with software to design hardware, so I might not be the right person for that.
But, perhaps, if you leave the issue open, someone in the community has the skills to do it.

Thanks!

stapelberg, 2015-12-30:
I think you are making the assumption that freetserv gets plugged into the front of the rack, i.e. the side where all the server LEDs are. The idea is to plug it into the back of the rack, though, where all the serial console ports, network cables and power cables are. Hence, freetserv’s front side is in the rack’s back side. I can add an actual picture of our rack in the next weeks.