freetserv/freetserv #40: PCB fab details

rich86, 2016-06-15:
I have never had a PCB fabbed before, nor do I have the requisite software to open and view the files, for people like me, it would be helpful if there could be a little bit more hand-holding whilst ordering from from Smart Prototyping - I haven't yet placed an order because I'm not sure of which options I need to change.

A screenshot or a list of options on the materials.html would be very useful.

Also, are there any obvious gotchas that people should be aware of if they order from another fab, such as OSH Park?

[image]

stapelberg, 2016-06-22:
Thanks for your interest in the project.

> I have never had a PCB fabbed before, nor do I have the requisite software to open and view the files, for people like me, it would be helpful if there could be a little bit more hand-holding whilst [...]

If you have never had a PCB fabbed, I have to ask: are you aware that SMD soldering is not a trivial task? If you haven’t done it before, you should definitely get a bunch of starter kits and train. The critical components on freetserv are 4 bucks a piece (for the FTDI), and messing up one of them will very likely render the entire board unuseable, which is a big waste of money and source of frustration :).

With that out of the way, here are the parameters I used:

```
PCB Prototyping 
  - Quantity: 10 
  - Max X-Dimension (cm): 10 
  - Max Y-Dimension (cm): 10 
  - Layers: 2 
  - PCB Thickness (mm): 1.6 
  - Copper Thickness: 1 oz 
  - Surface Finish: HASL (Hot Air Solder.. 
  - Solder Mask Color: Green 
  - E-Test Pass: 100% 
  - Solder Paste Stencil: no
  - Separated Sub-Boards: no 
  - Lead Time: Normal 
  - Gerber Files(zip or rar): gerber.zip
```

So, everything left to the default setting, except for the E-test being raised to 100% and the X/Y dimension adjusted because our PCB is bigger than 5cm x 5cm.

Hope that helps.

paravoid, 2020-12-20:
Piggybacking on this issue… a good option would be to use JLCPCB's SMT assembly service. I spot-checked a lot of the ICs and they all seem to be in their SMT parts library. The exact part numbers for some of the more standard components (capacitors etc.) don't match, but of course replacements exist. It shouldn't be too hard to create a BOM!

Note that they currently don't support connectors (like USB) but that's supposedely imminent. They also do not support through-hole components, but these are also easier to solder. Just having all the SMT components sourced and soldered would lower the bar here considerably IMHO :)

paravoid, 2020-12-20:
So I looked into it a bit more: 5 components (incl. the FT4232 and MAX3232) are available and 13 have a direct replacement from another vendor (same characteristics/package). On the flip side, 4 don't have a direct replacement (ferrite beads, 6MHz crystal, resistor array, HC-49/US inductor), 5 are through-hole components (RJ45 jacks, and board-to-board connectors) that are not supported but are also easy to solder, plus one SMT connector (microUSB) that is rumoured to be soon available for assembly.

More importantly though… JLCPCB's service doesn't support SMT assembly on both sides of the PCB, and the current design has the components almost equally split between the two sides as far as I can tell. So unless they add support for two-sided assembly, or someone is willing to go through the pain of a PCB redesign here, I don't see a lot of value in pursuing this further…