freetserv/freetserv #46: udev rules not working on Raspi 4

r-schmidt, 2020-05-15:
On the Raspi 2 the USB ports were identified as platform-bcm2708_usb-usb-[0-9] in freetserv.rules.
The Raspi 4 is a different platform so that fails. Instead the name is now platform-fd500000.pcie-pci-0000:01:00.0-usb-[0-9].

I made a little script that creates the rules file from a template:

`#!/bin/sh`

`rules_template_file="/usr/local/freetserv/freetserv.rules.template"`
`rules_file="/etc/udev/rules.d/freetserv.rules"`

`# create udev rules based on template file:`
`# find the ID of the USB port the PCBs are connected to`
`pcb_usb_port=$(udevadm test-builtin path_id /sys/class/tty/ttyUSB0 | grep -Po 'ID_PATH=\K.*usb-[0-9]?')`

`sed "s/FREETSERV_USB_PORT/$pcb_usb_port/g" "$rules_template_file" > "$rules_file"`

` # trigger update of the device name symlinks` 
` for i in $(seq 0 47); do udevadm test $(udevadm info -q path -n /dev/ttyUSB$i) >/dev/null 2>&1; done` 

The template is simply the existing freetserv.rules file where `platform-bcm2708_usb-usb-[0-9]`
if replaced with `FREETSERV_USB_PORT` which the script then replaces with the actual port.

stapelberg, 2020-05-15:
Thanks for filing this. Could you send a pull request which updates https://github.com/freetserv/freetserv/blob/05c33bea87f1ace5e6dc063f59996f533559ec67/udev/freetserv.rules please? I don’t mind duplication in there, and the file changes rarely.

By the way: did you build a freetserv instance? How is everything working out for you?

stapelberg, 2020-05-15:
This is fixed with pull request PR #47

closed 2020-05-15 by stapelberg

r-schmidt, 2020-05-15:
I'm totally new to Github so I hope I did the pull request right (or not completely wrong...).

My freetserv instance is quite different because the use case is very different:
I have shelves (with line cards) with serial ports from every slot combined into ribbon cables (12 slots per cable) for debugging.
So the PCB was modified with a pin header for the ribbon cable which connects to the different RJ45 soldering points and the RJ45 sockets are not assembled.
The front plate only contains one RJ45 socket for the Ethernet of the Raspi, a LED to see that power is on, power connector and a power switch.
The back plate has openings for 4 ribbon cables so I can connect 2 fully equipped shelves (48 slots).
The case is a really cheap plastic one: https://www.reichelt.de/19-rack-gehaeuse-1-he-abs-schwarz-rnd-455-00114-p193326.html?&trstct=pol_1&nbc=1
The front and back plate are replaced with 3D-printed ones for the connectors and switch (could also make holes in the existing plates but printing is cool...).

On the raspi I used raspian (Debian is not there yet for proper out of the box use on raspi 4).
There is a single passwordless user that all telnet connections log into. The serial port is selected by giving the name to telnet as user name (parameter -l) instead of the port number like in the commercial terminal servers.
I had to change /etc/pam.d/common-auth to allow login without password (nullok instead of nullok_secure).

Using SSH would have meant to provide everyone with the same private key or add many public keys to the user. Or have many users like what you use which is also not feasible for us.
It is a development tool and which slot is used by whom can change multiple times per day and there are also multiple users looking at the same port.

After login the provided username is used to select which screen instance to attach to. This makes management really simple.
And it allows to use aliases for the ports without having to add more symlinks via udev.
I'm not completely done with that yet (e.g. managing aliases with a simple management interface is on the todo list).

stapelberg, 2020-05-15:
Sounds great, thanks for sharing! Do you have any pictures? I’m glad to learn that freetserv was useful for your project.

Your pull request was fine, too! :)