freetserv/freetserv #48: using more than 34 ports at the same time crashes the kernel

r-schmidt, 2020-12-16:
This is obviously a bug/limitation in Linux but I just wanted to mention it here.
I'm using screen to access the serial ports and leave the sessions running so I can re-attach later.
Now I saw that the moment I create the 35th session, the kernel crashes.
It doesn't matter which port the session is created for.

I'm using the latest Raspberry Pi OS on a Raspi 4B with 2GB RAM and the issue is reproducible every time and on all of my 48 port freetservs.

I was just wondering if you also see that on your system that unless you changed it is running a Debian?

stapelberg, 2020-12-16:
Thanks for documenting this issue.

I haven’t run into this myself because we never had that many sessions on freetserv.

Perhaps you can raise this on the Linux kernel mailing list and see if anyone can help there?

raylee, 2020-12-16:
@r-schmidt could you post the kernel's oops message from the crash? Even an image would be helpful, though the complete oops in text form is preferred. You'll want that for posting to the linux kernel mailing list regardless.

r-schmidt, 2020-12-17:
Ok, here is what is coming on the console (there is more coming every ~30 seconds but that is probably irrelevant).
I haven't tried connecting the ports to a different computer (e.g. my notebook) to see if it happens there as well.
Will maybe do that later today.

[  103.715803] 8<--- cut here ---
[  103.720089] Unhandled fault: asynchronous external abort (0x1211) at 0x00000000
[  103.728619] pgd = d30b99fe
[  103.732532] [00000000] *pgd=17787003, *pmd=00000000
[  103.738634] Internal error: : 1211 [#1] SMP ARM
[  103.744388] Modules linked in: sha256_generic libsha256 cfg80211 rfkill 8021q garp stp llc ftdi_sio usbserial vc4 cec rpivid_mem v3d drm_kms_helper gpu_sched drm bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_isp(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common drm_panel_orientation_quirks raspberrypi_hwmon snd_soc_core videodev mc vc_sm_cma(C) snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops uio_pdrv_genirq uio ip_tables x_tables ipv6
[  103.797884] CPU: 0 PID: 2336 Comm: sudo Tainted: G         C        5.4.79-v7l+ #1373
[  103.807232] Hardware name: BCM2711
[  103.812161] PC is at lockref_put_return+0x50/0x9c
[  103.818379] LR is at dput.part.8+0x58/0x3b8
[  103.824052] pc : [<c06a2f98>]    lr : [<c04287c0>]    psr: 60000013
[  103.831839] sp : d6dbbe50  ip : 00000000  fp : d6dbbe6c
[  103.838602] r10: d71f8c70  r9 : 00000001  r8 : 00080060
[  103.845375] r7 : 00000001  r6 : 00030003  r5 : 00000000  r4 : 00030003
[  103.853475] r3 : 00000001  r2 : 00030003  r1 : 00000064  r0 : d91b1fb8
[  103.861570] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  103.870286] Control: 30c5383d  Table: 177ea000  DAC: fffffffd
[  103.877630] Process sudo (pid: 2336, stack limit = 0xb7a13c6b)
[  103.885085] Stack: (0xd6dbbe50 to 0xd6dbc000)
[  103.891075] be40:                                     d91b1f68 d91b1fb8 c0de3628 00080040
[  103.900914] be60: d6dbbea4 d6dbbe70 c04287c0 c06a2f54 00000000 c040e3bc 00000000 d6c58180
[  103.910762] be80: d71f8c70 080a801d ef32e850 d91b1f68 00000010 d71f8c70 d6dbbeb4 d6dbbea8
[  103.920635] bea0: c0428b38 c0428774 d6dbbef4 d6dbbeb8 c040e42c c0428b2c 00000000 00000000
[  103.930523] bec0: c020ce54 d6c58188 d6d0eff0 d6c58180 00000000 d6d0eac0 d6d0f01c c12b9d54
[  103.940423] bee0: d6d0eff0 c1204fc8 d6dbbf04 d6dbbef8 c040e59c c040e34c d6dbbf2c d6dbbf08
[  103.950351] bf00: c02433f8 c040e590 ffffe000 d6dbbfb0 c02011c4 00000004 fffffe30 5ac3c35a
[  103.960296] bf20: d6dbbfac d6dbbf30 c020ce54 c0243350 d6dbbf4c d6dbbf40 5ac3c35a c1204fcc
[  103.970269] bf40: d6dbbf6c d6dbbf50 c0407de8 c040eb18 00000004 d6d60600 d7a73c00 00000000
[  103.980258] bf60: d6dbbf94 d6dbbf70 c042fcc0 c0407d84 02008150 b6e63b1c 00000000 8dd7b910
[  103.990263] bf80: c02011c4 02008150 b6e63b1c 00000000 00000006 c02011c4 d6dba000 00000006
[  104.000294] bfa0: 00000000 d6dbbfb0 c0201034 c020c8cc 00000000 00000444 b6e636d8 b6d83864
[  104.010348] bfc0: 02008150 b6e63b1c 00000000 00000006 b66ff29c b66ff298 020082b8 00000000
[  104.020416] bfe0: 00000000 bef0ef68 b6d85088 b6de8f74 20000010 00000004 00000000 00000000
[  104.030494] Backtrace:
[  104.034839] [<c06a2f48>] (lockref_put_return) from [<c04287c0>] (dput.part.8+0x58/0x3b8)
[  104.044880]  r7:00080040 r6:c0de3628 r5:d91b1fb8 r4:d91b1f68
[  104.052509] [<c0428768>] (dput.part.8) from [<c0428b38>] (dput+0x18/0x1c)
[  104.061293]  r10:d71f8c70 r9:00000010 r8:d91b1f68 r7:ef32e850 r6:080a801d r5:d71f8c70
[  104.071136]  r4:d6c58180
[  104.075682] [<c0428b20>] (dput) from [<c040e42c>] (__fput+0xec/0x244)
[  104.084163] [<c040e340>] (__fput) from [<c040e59c>] (____fput+0x18/0x1c)
[  104.092917]  r10:c1204fc8 r9:d6d0eff0 r8:c12b9d54 r7:d6d0f01c r6:d6d0eac0 r5:00000000
[  104.102826]  r4:d6c58180
[  104.107432] [<c040e584>] (____fput) from [<c02433f8>] (task_work_run+0xb4/0xd4)
[  104.116843] [<c0243344>] (task_work_run) from [<c020ce54>] (do_work_pending+0x594/0x59c)
[  104.127050]  r9:5ac3c35a r8:fffffe30 r7:00000004 r6:c02011c4 r5:d6dbbfb0 r4:ffffe000
[  104.136925] [<c020c8c0>] (do_work_pending) from [<c0201034>] (slow_work_pending+0xc/0x20)
[  104.147255] Exception stack(0xd6dbbfb0 to 0xd6dbbff8)
[  104.154468] bfa0:                                     00000000 00000444 b6e636d8 b6d83864
[  104.164831] bfc0: 02008150 b6e63b1c 00000000 00000006 b66ff29c b66ff298 020082b8 00000000
[  104.175197] bfe0: 00000000 bef0ef68 b6d85088 b6de8f74 20000010 00000004
[  104.184017]  r10:00000006 r9:d6dba000 r8:c02011c4 r7:00000006 r6:00000000 r5:b6e63b1c
[  104.194070]  r4:02008150
[  104.198818] Code: 1a000002 e1a0cf94 e33c0000 1afffff8 (e1570003)
[  104.207145] ---[ end trace e5ed96736c0aa23a ]---
[  104.214070] pcieport 0000:00:00.0: AER: Multiple Uncorrected (Non-Fatal) error received: 0000:00:00.0
[  105.586945] ------------[ cut here ]------------
[  105.593927] WARNING: CPU: 2 PID: 131 at drivers/firmware/raspberrypi.c:63 rpi_firmware_transaction+0xec/0x128
[  105.606193] Firmware transaction timeout
[  105.606196] Modules linked in: sha256_generic libsha256 cfg80211 rfkill 8021q garp stp llc ftdi_sio usbserial vc4 cec rpivid_mem v3d drm_kms_helper gpu_sched drm bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_isp(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common drm_panel_orientation_quirks raspberrypi_hwmon snd_soc_core videodev mc vc_sm_cma(C) snd [... cut ...]
[  273.524921] [<c0245e60>] (kthread) from [<c02010ac>] (ret_from_fork+0x14/0x28)
[  273.533763] Exception stack(0xef911fb0 to 0xef911ff8)
[  273.540424] 1fa0:                                     00000000 00000000 00000000 00000000
[  273.550232] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  273.560026] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[  273.568242]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c0245e60
[  273.577668]  r4:ef86fcc0

r-schmidt, 2020-12-17:
Testing with the notebook was easier than thought. There everything works fine.
So it seems related to the Raspi. According to the console output it is either the PCIe driver or the way the Raspi firmware handles it.

stapelberg, 2020-12-17:
Good to know!

So, one possible workaround might be to replace the Raspberry Pi with e.g. a https://rockpi.org/ or other form-factor compatible (or close) single board computers.

Of course, it would be good to fix the issue with the Raspberry Pi, I’m just mentioning this in case you need to get something working with little time investment.