freetserv/freetserv #49: Mention Commercial Options

lathama, 2021-04-28:
I was about to build one and found a commercial option from Coolgear Inc. with model number USB-32COM-RM that I have on order. It may be an option for those that do not want to build as the costs are close to the BOM

stapelberg, 2021-04-28:
Thanks for the pointer!

Searching for the model number without the vendor yields a similar device at https://www.vscom.de/vscom-usb-32com-rm.html, which is phased out however. Not sure how long the Coolgear one has been around/will be around :)

I’m curious what you think about the Coolgear device once you have it.

I originally bought a commercial one second-hand, too, but was disappointed with hardware reliability (device wouldn’t work well) and software quality (terribly old stack, unmaintained).

lathama, 2021-04-29:
I will report back after some testing. Has not arrived yet, trying to think about how to load test it.

lathama, 2021-05-05:
Arrived, here is what it looks like from lsusb
```
Bus 003 Device 018: ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
Bus 003 Device 017: ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
Bus 003 Device 015: ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
Bus 003 Device 013: ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
Bus 003 Device 016: ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
Bus 003 Device 014: ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
Bus 003 Device 012: ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
Bus 003 Device 011: ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
```

lathama, 2021-05-05:
Pictures at https://lathama.net/USB-32COM-RM

lathama, 2021-05-06:
Works well, trying to figure out how the ports got loaded in order because that is awesome.

stapelberg, 2021-05-09:
Can you run lsusb.py (part of the lsusb package on some distributions) as well to show the topology?

Are you planning to use the freetserv Raspberry Pi setup with this device, then?

lathama, 2021-05-09:
`
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/5p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 5: Dev 53, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 05e3:0610 Genesys Logic, Inc. Hub
        |__ Port 1: Dev 54, If 0, Class=Hub, Driver=hub/4p, 480M
            ID 05e3:0610 Genesys Logic, Inc. Hub
            |__ Port 1: Dev 56, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
                ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
            |__ Port 1: Dev 56, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
                ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
            |__ Port 1: Dev 56, If 3, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
                ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
            |__ Port 1: Dev 56, If 1, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
                ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
            |__ Port 2: Dev 58, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
                ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
            |__ Port 2: Dev 58, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
                ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
            |__ Port 2: Dev 58, If 3, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
                ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
            |__ Port 2: Dev 58, If 1, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
                ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
            |__ Port 3: Dev 60, If 3, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
                ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
            |__ Port 3: Dev 60, If 1, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
                ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
            |__ Port 3: Dev 60, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
                ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
            |__ Port 3: Dev 60, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
                ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
            |__ Port 4: Dev 62, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
                ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
            |__ Port 4: Dev 62, If 1, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
                ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
            |__ Port 4: Dev 62, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
                ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
            |__ Port 4: Dev 62, If 3, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
                ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
        |__ Port 2: Dev 55, If 0, Class=Hub, Driver=hub/4p, 480M
            ID 05e3:0610 Genesys Logic, Inc. Hub
            |__ Port 3: Dev 61, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
                ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
            |__ Port 3: Dev 61, If 3, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
                ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
            |__ Port 3: Dev 61, If 1, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
                ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
            |__ Port 3: Dev 61, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
                ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
            |__ Port 1: Dev 57, If 1, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
                ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
            |__ Port 1: Dev 57, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
                ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
            |__ Port 1: Dev 57, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
                ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
            |__ Port 1: Dev 57, If 3, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
                ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
            |__ Port 4: Dev 63, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
                ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
            |__ Port 4: Dev 63, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
[... cut ...]
            |__ Port 2: Dev 59, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
                ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
            |__ Port 2: Dev 59, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
                ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
`

Currently controlled by a PC Engines APU2. When I opened the case I noted there is room to install a SBC inside.

lathama, 2021-05-09:
https://lathama.net/USB-32COM-RM#Software_Setup for the software setup I just did. Turned out to be simpler than I thought

stapelberg, 2021-05-14:
Mentioned this (including links to the product and your notes) on the freetserv github page.

If you think anything else should be mentioned/changed, please submit a pull request with your suggestion.

Thank you!

closed 2021-05-14 by stapelberg

lathama, 2021-05-14:
Thanks, I will keep an eye out for other options and some day get the cycles to build my own